squix78/json-streaming-parser #3: ExampleListener does not name a type

llpereiras, 2016-02-04:
fatal error: ExampleParser.h: No such file or directory
 #include <ExampleParser.h>
                           ^
compilation terminated.
exit status 1
Error compiling.

Blinkesp8266_ftdi_baterry:13: error: 'ExampleListener' does not name a type
 ExampleListener listener;
 ^
/media/leandro/Disco Local/ArduinoProjetos/Blinkesp8266_ftdi_baterry/Blinkesp8266_ftdi_baterry.ino: In function 'void setup()':
Blinkesp8266_ftdi_baterry:24: error: 'listener' was not declared in this scope
   parser.setListener(&listener);
                       ^
exit status 1
'ExampleListener' does not name a type

maxpowel, 2016-08-22:
This is because you didnt copy the file ExampleParser.h

llpereiras, 2016-08-23:
@maxpowel I will check this. Thanks!

mrohner, 2017-01-22:
This is not about an issue per see but lack of understanding the example. May be can give me a hint on how to proceed. 
This is the link to the json I'm trying to parse:
http://online.fahrplan.zvv.ch/bin/stboard.exe/dny?input=Z%C3%BCrich,+Oerlikon+(SBB)&dirInput=Z%C3%BCrich+HB&maxJourneys=6&boardType=dep&start=1&tpl=stbResult2json

How do I translate this to the structure given in the example:
char json[] = "{\"a\":3, \"b\":{\"c\":\"d\"}}";

Any idea?