squix78/json-streaming-parser #8: Problem in understanding the example

mrohner, 2017-01-22:
This is not about an issue per see but lack of my understanding the example. May be you can give me a hint on how to proceed.
This is the link to the json I'm trying to parse:
http://online.fahrplan.zvv.ch/bin/stboard.exe/dny?input=Z%C3%BCrich,+Oerlikon+(SBB)&dirInput=Z%C3%BCrich+HB&maxJourneys=6&boardType=dep&start=1&tpl=stbResult2json

How do I translate this to the structure given in the example:
char json[] = "{"a":3, "b":{"c":"d"}}";

Any idea?

squix78, 2017-01-23:
Hi
I'm currently working on a blog post which will be published this Friday about how to use the parser. You can see the draft in the attachment. Let me know if that helped or what is unclear...[
[ESP8266_ The JSON Streaming Parser _ Squix - TechBlog.pdf](https://github.com/squix78/json-streaming-parser/files/723785/ESP8266_.The.JSON.Streaming.Parser._.Squix.-.TechBlog.pdf)
](url)

mrohner, 2017-01-23:
Hallo Daniel,

Danke für das Dokument. Woher nimmst Du die City, scheint ja gar nicht in Deinem JSON stream vorhanden zu sein.

Nein, leider verstehe ich's immer noch nicht. Wie kann man Objekte überspringen? In meinem Datenstrom kommen die selben Keys immer wieder vor (brauche sie aber nicht). Ich hab die für mich relevanten Felder rot markiert.

Ich verstehe, dass ich ein eigenes .CPP erstellen soll.

Jetzt will nach dem key  "name" suchen, aber nicht das gelb markierte, sondern das rote. Wie spring ich jetzt an diese Position?

Dann brauch ich zwei Mal den key "countdown" und dann lange nichts mehr bis zum nächsten "product".

Wie springe ich zum nächsten Objekt? Kann ich dann jezt endObject(); aufrufen oder endArray(); oä.?

Danke für Deine Hilfe

{
"station":{
"name":"Z&#252;rich Oerlikon (SBB)",
"additionalInformation":{
"attr_MU": false,
"attr_HU": true,
"attr_HU_val": "13034"
,"timetableUrl":"http://www.zvv.ch/zvv/de/fahrplan/haltestellenfahrplaene.html?hstNr=13034"
}
},
"connections":[
{
"name":"0",
"date":"23.01.17",
"product":{ "name":"Trm   14", "longName":"Tram", "type":"9" ,"line":"14" ,"direction":"Z&#252;rich, Triemli" ,"directionType":"to" ,"icon":"icon_tram" ,"color":{"fg":"ffffff","bg":"008dc5"} }
,
"mainLocation":{
"location":{
"name":"Z&#252;rich, Bahnhof Oerlikon Ost",
"x":8545847,
"y":47413334,
"id":"299598",
"tupelId":"",
"type":"STATION"
},
"time":"16:14",
"countdown":"58",
"date":"23.01.17",
"platform":"",
"realTime": {
"time":"",
"countdown":"",
"date":"",
"delay":"",
"platform":"",
"isPlatformChanged":false,
"isDelayed":true,
"hasRealTime":false
}
},
"locations":[
{
"location":{
"name":"Z&#252;rich, Bahnhof Oerlikon Ost",
"x":8545847,
"y":47413334,
"id":"299598",
"tupelId":"",
"type":"STATION"
},
"time":"16:14",
"countdown":"58",
"date":"23.01.17",
"platform":"",
[... 1684 lines cut ...]
"realTime": {
"time":"",
"countdown":"",
"date":"",
"delay":"",
"platform":"",
"isPlatformChanged":false,
"isDelayed":true,
"hasRealTime":false
}
}
],
"trainInfo":"601617/203712/923736/261330/85",
"cancelled":false,
"attributes_bfr":[
],
"attributes":[
],
"himMessages":[
],
"hasHIMMessage":false,
"hasGlobalHIMMessage":false
}
]
}

squix78, 2017-01-23:
I will answer in English, then more people can profit: if your keys are repeating you'll have to remember your context. You could for instance remember the name of the parent of your current object. You do this in the startObject method and remove it in the endObject method. Have a look here:
https://github.com/squix78/esp8266-weather-station/blob/master/WundergroundClient.cpp#L195
https://github.com/squix78/esp8266-weather-station/blob/master/WundergroundClient.cpp#L494

The second countdown key/value pair for instance has the parent "realtime". So if you were interested in realtime.countdown value you would check if(currentParent == "realtime" && currentKey == "countdown")

If that doesn't help tell me which attributes you need and I will try to hack together the class for you...

mrohner, 2017-01-30:
Hi,
I gave it a try and even though I still do not fully understand the logic I achieved some promising results.
Nevertheless some questions came up (above JSON data is parsed) 
I get values for "type" and "line" but NOT for the two countdown fields. Any ideas?
Is it ok to 'for' loop to capture ten incidents?
Is the parent key retained (e.g. "mainLocation") when a subsequent parent (e.g. "location") overwrites the CurrentParent value? The countdown field is still in "mainLocation" but currentParent has already changed to "location"?
Thank you

void ZVVClient::value(String value) {
  for (int i = 0; i < 9 ; i++) { //Number_Journeys
    if (currentParent == "product") { 
      if (currentKey == "type") {
        type[i] = value.toInt();
        }
      if (currentKey == "line") {
        line[i]  = value.toInt();
        }
    }
    if (currentKey == "countdown") {
      a_countdown[i]  = value.toInt();
    }
    if (currentParent == "realTime" && currentKey == "countdown") { 
      countdown[i]  = value.toInt();
      }
    } 
  }

mrohner, 2017-01-31:
I found a solution myself by tracking all the hierarchy levels of the object. Thanks
void ZVVClient::value(String value) {
  //Serial.print(currentParent[hierarchy_level]);
  //Serial.print("   ");
  //Serial.println(currentKey);
    if (currentParent[hierarchy_level] == "product") {  // Has a Parent key and 2 sub-keys
      if (currentKey == "type") {
        count++;
        Serial.println(count);
        type[count] = value.toInt();
        //Serial.println(type[count]);
        }
      if (currentKey == "line") {
        line[count]  = value.toInt();
        //Serial.println(line[count]);
        }
    }
    if (currentParent[hierarchy_level] == "mainLocation" && currentKey == "countdown") {
      a_countdown[count]  = value.toInt();
      //Serial.println(a_countdown[count]);
      }
      if (currentParent[hierarchy_level-1] == "mainLocation" && currentParent[hierarchy_level] == "realTime" && currentKey == "countdown") { 
      countdown[count]  = value.toInt();
      //Serial.println(countdown[count]);
    }
 }

void ZVVClient::endArray() {
}

void ZVVClient::startObject() {
  hierarchy_level++;
  currentParent[hierarchy_level] = currentKey;
}

void ZVVClient::endObject() {
  hierarchy_level--;
}

closed 2017-01-31 by mrohner